eez-open/studio #327: [LVGL] ScreensEnum misaligned?

mdkendall, 2024-03-07:
LVGL project without Flow.  When a button is pressed I wish to show a certain screen. So I have attached an event handler to the button's `PRESSED` action; it calls a user action named `showScreenMain`. In my code I write the appropriate function, and it seems to me I must call `loadScreen(index)` with the index of the screen I wish to show.

```
void action_show_screen_main(lv_event_t * e) {
    loadScreen(0);
}
```

But how to know what this index is? I can see it should be `0` because it is clear that is an index into `objects_t objects`, and `main` is the first field in that struct. But I should not have to know this. I look in `screens.h` and I see a useful enumeration:

```
enum ScreensEnum {
    SCREEN_ID_MAIN = 1,
    SCREEN_ID_TWO = 2,
};
```

But it seems to me this is off-by-one. The enumeration starts at `1`, but the indexes I need to pass to `loadScreen()` start at `0`.

Do I just have to subtract `1` from the enumerated values? Have I misunderstood the purpose of the `ScreensEnum` enumeration? Is there some other way to know the indexes I must pass to `loadScreen()`?

mvladic, 2024-03-08:
You can safely pass members from `ScreensEnum` subtracted by 1. 

However, I confess, this is not elegant and it should be fixed so that `loadScreen` accepts `ScreenEnums`. 

So, `ui.h` file template should be:

```
#ifndef EEZ_LVGL_UI_GUI_H
#define EEZ_LVGL_UI_GUI_H

#include "lvgl/lvgl.h"
#include "screens.h"

void ui_init();
void ui_tick();

void loadScreen(ScreensEnum screenId);

#endif // EEZ_LVGL_UI_GUI_H
```

and `ui.c`:

```
#include "ui.h"
#include "images.h"

static int16_t currentScreen = -1;

static lv_obj_t *getLvglObjectFromIndex(int32_t index) {
    if (index == -1) {
        return 0;
    }
    return ((lv_obj_t **)&objects)[index];
}

static const void *getLvglImageByName(const char *name) {
    for (size_t imageIndex = 0; imageIndex < sizeof(images) / sizeof(ext_img_desc_t); imageIndex++) {
        if (strcmp(images[imageIndex].name, name) == 0) {
            return images[imageIndex].img_dsc;
        }
    }
    return 0;
}

void loadScreen(ScreensEnum screenId) {
    currentScreen = screenId - 1;
    lv_obj_t *screen = getLvglObjectFromIndex(currentScreen);
    lv_scr_load_anim(screen, LV_SCR_LOAD_ANIM_FADE_IN, 200, 0, false);
}

void ui_init() {
    create_screens();
    loadScreen(0);
}

void ui_tick() {
    tick_screen(currentScreen);
}
```

In the next release this will be changed in the [LVGL template](https://github.com/eez-open/studio/blob/master/resources/project-templates/LVGL-8.3.eez-project). In the meantime you can make these changes by yourself for each project. File templates can be edited in the Settings:

[image]

mdkendall, 2024-03-08:
Great, thank you for your help.

mdkendall, 2024-03-08:
Small correction. The declaration of `loadScreen` must be
```
void loadScreen(enum ScreensEnum screenId);
```
because `ScreensEnum` is a tag, not a typedef name. Otherwise, the solution works.

closed 2024-03-11 by prasimix