eez-open/studio #328: Additional Fonts not rendering correctly with LVGL v9 in EEZ Studio v0.11.0

janbernrader, 2024-03-15:
When opting for the LVGL v9 version there seems to be an issue with the fonts rendering, see the image attached even though the characters display correctly in the font manager.

[image]
[image]

Furthermore there seem to be some symbols missing with the keyboard widget.
[image]
[image]

mvladic, 2024-03-15:
For some reason 8 bits per pixel font doesn't work in LVGL 9.0, so choose 4 bits per pixel when creating font.

mvladic, 2024-05-03:
When adding a font in LVGL project for version 9.0, the default value for bits per pixel will be set to 4 bits.

mvladic, 2024-05-06:
Done.

closed 2024-05-06 by mvladic