eez-open/studio #335: EEZ v0.11.0 / LVGL 9.x - Spinner Widget compiling error - inexistant function

anvgfr, 2024-04-06:
Hello,

When creating a project targeting LVGL 9.0, and using spinner, project compile failed since EEZ use a non-existent function to create the widget :

in generated code :
```
            lv_obj_t *obj = lv_spinner_create(parent_obj, 1000, 60);
            objects.wifi_sp_scanning = obj;
```
But in LVGL widget there is only the function:
```
/**
 * Create a spinner widget
 * @param parent    pointer to an object, it will be the parent of the new spinner.
 * @return          the created spinner
 */
lv_obj_t * lv_spinner_create(lv_obj_t * parent);
```
Need to change generated code to do things in two steps : 

```
          lv_obj_t *obj=lv_spinner_create(parent_obj);
          lv_obj_set_size(obj,1000,60);
          objects.wifi_sp_scanning = obj;
```

mvladic, 2024-04-07:
Fixed.

anvgfr, 2024-04-07:
thanks.  good job!

closed 2024-04-08 by prasimix