eez-open/studio #336: Lack of define in ui.h

anvgfr, 2024-04-09:
Hello,

Compiling ui with arduino ide 2.0, there is some define that are lacking : (got a linker error)
- extern "C" defines for ui.h

```
#ifndef EEZ_LVGL_UI_GUI_H
#define EEZ_LVGL_UI_GUI_H

#include <lvgl.h>
#include "screens.h"

void ui_init();
void ui_tick();

void loadScreen(enum ScreensEnum screenId);

#endif // EEZ_LVGL_UI_GUI_H
```

good one should be :

```
#ifndef EEZ_LVGL_UI_GUI_H
#define EEZ_LVGL_UI_GUI_H

#include <lvgl.h>
#include "screens.h"

#ifdef __cplusplus
extern "C" {
#endif

void ui_init();
void ui_tick();

void loadScreen(enum ScreensEnum screenId);

#ifdef __cplusplus
}
#endif

#endif // EEZ_LVGL_UI_GUI_H
```

mvladic, 2024-04-09:
Thanks!

It will be added in the template from which new LVGL projects are created.

mvladic, 2024-04-10:
Fixed.

Since templates are downloaded from https://github.com/eez-open/eez-project-templates repository, this fix is already in effect with the latest released version of Studio (v0.12.0).

closed 2024-04-14 by anvgfr