eez-open/studio #342: Dashboard Line Chart widget: dynamically add a line

fietser28, 2024-04-14:
Some measurements are creating multiple lines in a dynamic way. Common examples: 
- Discharging a battery for a range of discharge rates. It is not known during programming how many lines will be created.
- DC-DC converter efficiency depending on both input voltage and output current. 

I think a few possible solutions to support this is are possbile:
- add a checkbox 'multiple lines' to a single line configuration. If this checkbox is checked an additional Flow input will be created. If an event into this input evaluates to true, a new line is dynamically added an data going into this line will be send to this new line
- add an expression field. If the value evaluates to true a new line is created. This allows the user to be more flexible but might also be hard to get the events right (and not create unwanted lines)?

mvladic, 2024-04-14:
I think by adding "Visible" expression property to each line would be solution for this.

fietser28, 2024-04-14:
With the "Visible" do you mean: 
1. A fixed set of lines is predefined during flow development by the user. Possibly (but not required) with all fields equal. If this is the case data is feed to all lines, but using an expression in the "Visible" field data is only excepted for that line if "Visible" evaluates to true. This is flexible but not completely dynamic. The advantage is the colours can be set manually. 
If Visible is never true during runtime for a specific, the line and it's legend entry are not visible.

OR 

2. Do you mean the expression for visible evaluates to a number and the data entered will be added to the dynamically created line corresponding that number. This is truly dynamic, supporting a large amount of lines, but requires some algorithm to choose colours (from a palette?).

Or something else?

mvladic, 2024-04-14:
I mean option 1.