eez-open/studio #343: Support For Windows 7 ?

KASSIMSAMJI, 2024-04-15:
Here There, I just downloaded the latest version of EEZ Studio (EEZ-Studio-Setup-0.12.0 ) and here what I am getting when trying to run it

[image]

Downgraded to ver: 1.0.0 , still same error

I am using Windows 7 Ultimate with Service Pack 1

Any Help is highly appreciated

mvladic, 2024-04-15:
EEZ Studio is using [Electron](https://www.electronjs.org/) framework and [Electron from version 23 doesn't support Windows 7](https://www.electronjs.org/blog/windows-7-to-8-1-deprecation-notice). We are currently using version 28.1.0, so that's the reason why it doesn't work.

KASSIMSAMJI, 2024-04-15:
Thank you for clarification

closed 2024-04-15 by KASSIMSAMJI