eez-open/studio #344: Dashboard history

fietser28, 2024-04-21:
Just like a SCPI instrument I would like the possibility to give a dashboard a history for logging output of dashboard runs. The history should have the same features (add note, export to notebook, delete, pause) as the current instrument log.

It should log the runs and output generated by the dashboard:
It will require an addition widget 'AddToDashboardHistory' where text, charts, csv (?), json(?) and images can be logged to the history. It is also nice to have the option to add comments to results to give context.

mvladic, 2024-04-21:
I think we should use Notebooks for this. Similar to InstrumentObject type, we can add NotebookObject type with following actions:

- CreateNotebook: creates a new notebook with given name. For example, a name can be dashboard file name.
- SelectNotebook: opens dialog box to select notebook
- GetNotebook: get existing notebook by name

Once we have variable holding reference to Notebook we can use AddToHistory action to append text/image/chart/,,, to its history.  This means we don't need AddToInstrumentHistory action anymore (it will be automatically migrated to AddToHistory action), because target object for AddToHistory can be either instrument or notebook.

This requires no new concept (dashboard history) in Studio. We already have a way to manage notebooks and browse through notebook history.

mvladic, 2024-04-21:
> It should log the runs and output generated by the dashboard

This will be left to the developer. He can add some text in notebook history each time when dashboard starts.