eez-open/studio #345: Dashboard screenshot action

fietser28, 2024-04-21:
I would like to have a way to have my dashboard generate screenshots (from a selected page or container of the dashboard?) so they can be saved to file or history (see #344).

I think a action to take the screenshot and output the image as a blob is the way to go. Inside the widget the user can select from a list of pages or containers the scope of the screenshot. I think containers are more flexible for this.

mvladic, 2024-04-21:
[Electron](https://www.electronjs.org/) framework, that we use in Studio, has a [capturePage(rectangle)](https://www.electronjs.org/) API - so, it should be possible to implement this feature. But, it will be limited in the way that dashboard page or selected container must be visible at the time when capture API is called.

fietser28, 2024-04-21:
I think that is fine for now: Just like physical instruments where a screen capture is a common feature.

Creating images from non visible containers is more like a report generation feature. Maybe somewhere in the future.