eez-open/studio #347: Var types incompatible in screen tick

Luke-Shepp, 2024-04-22:
My generated screens.c looks something like this;

```c++
void tick_screen_main() {
    [...trimmed...]
    {
        const char *new_val = get_var_speed();
        const char *cur_val = lv_label_get_text(objects.obj2);
        if (strcmp(new_val, cur_val) != 0) {
            tick_value_change_obj = objects.obj2;
            lv_label_set_text(objects.obj2, new_val);
            tick_value_change_obj = NULL;
        }
    }
    [....trimmed....]
}
```

`get_var_speed` is defined in the generated `vars.h` as:

```c++
extern float get_var_speed();
```

That is because I have chosen it to be a float value in the UI:

[image]

Due to the type mismatch, the error on build is:

```
ui/screens.c: In function 'tick_screen_main':
ui/screens.c:166:31: error: incompatible types when initializing type 'const char *' using type 'float'
  166 |         const char *new_val = get_var_speed();
      |                               ^~~~~~~~~~~~~
ui/screens.c: In function 'tick_screen_settings':
```

There's another warning for another variable, but it's not a hard error this one:

```
ui/screens.c:292:31: warning: initialization of 'const char *' from 'int32_t' {aka 'long int'} makes pointer from integer without a cast [-Wint-conversion]
  292 |         const char *new_val = get_var_wifi_state();
      |                               ^~~~~~~~~~~~~~~~~~
make: *** [ui/screens.o] Error 1
```

mvladic, 2024-04-22:
You are not using EEZ Flow, so conversion of variable types, in this case `float` to `string` (`const char *` in C) will not be done in generated code. That means, variable `speed` in your case must be of type `string` not `float` as Label widget is expecting string. But, we can say this is bug since there is no error reported for this case. Also, we should consider to put type conversion code in generated code in this case even if EEZ Flow is not used.

cwilling, 2025-02-17:
I've also come across the same problem - variable is stored as int32_t but `lv_label_set_text(objects.obj2, new_val);` wants `new_val` to be a string. On Arduino, the compiler doesn't bother mentioning a mismatch - just continues; however the uploaded code causes the device to constantly reboot. I can modify screens.c locally to fix my particular case but, of course, any local changes are over-written every time I ctrl-b in EEZ Studio.

mvladic, 2025-02-17:
For now, you should use string variable for the Label widget and make conversion from int32_t/float to string in the `get_var` function .

cwilling, 2025-02-18:
OK. It turns out that when I use a string variable for the Label widget, it's actually easier to leave vars.c the same as the code suggested by the eez-studio gui (which fits in with what is expected by generated code in screens.c) and make any string<->int32_t conversions in my actions.c file (where any label changes are instigated).