eez-open/studio #350: LVGL Meter Widget without name causes exception if has expression based needle

Luke-Shepp, 2024-04-22:
In a new LVGL project, create a float variable, lets say "X". Add a meter widget to the page, and add a needle line with the value of "X" Variable.

When trying to build, this fails with exception:

[image]

This can be resolved by adding a "name" to the meter widget. 

It looks like, `isAccessibleFromSourceCode` is false for a widget in this state, which means the object doesn't make it into `identifiersMap` (see `LVGLIdentifiers::enumIdentifiers`) so building the LVGL code for a child of this widget (the needle line) fails.

I suspect it just needs validation to ensure a name is set? 😃

mvladic, 2024-04-23:
This should be fixed now.

Luke-Shepp, 2024-04-23:
Yeah that's got it - thank you 😁

closed 2024-05-06 by mvladic