eez-open/studio #352: dashboard project: Not mandatory to connect the instrument

dong-king, 2024-04-30:
In version 0.12.0, when I open a  dashboard project, eez-studio forces me to connect the instrument, otherwise I cannot enter the debug interface.
Is it possible to provide an option to enter the dashboard project interface without connecting the instrument?

By the way, In version 0.12.0， View->Externsion Manager and View->Settings are not working.

mvladic, 2024-04-30:
> In version 0.12.0, when I open a dashboard project, eez-studio forces me to connect the instrument, otherwise I cannot enter the debug interface.
> Is it possible to provide an option to enter the dashboard project interface without connecting the instrument?

Only for variables marked as "Persistent", Studio will ask to connect to the instrument before starting a dashboard. So, if you don't want this you can uncheck this option.

[image]

> By the way, In version 0.12.0， View->Externsion Manager and View->Settings are not working.

Thanks, will be fixed.

mvladic, 2024-04-30:
BTW you can manually open select instrument dialog at any time with the SelectInstrument action component:

[image]

mvladic, 2024-05-03:
> By the way, In version 0.12.0， View->Externsion Manager and View->Settings are not working.

Fixed.

closed 2024-05-06 by mvladic