eez-open/studio #357: Syntax error in LVGL code for version 9.0 when Roller widget is used

mvladic, 2024-05-06:
Reported by @enrip19 on discord:

Also I found that when a roller is defined it has a line which is not completed when it creates a screen in screen.c. I can solve it replacing LVOBJ_FLAG by LV_OBJ_FLAG_CLICKABLE

[image]

enrip19, 2024-05-09:
I've stumbled upon some new issues with the roller, this time within the eezStudio GUI. As you can see in the attached picture, it seems that there's currently no option available to control the style of the selection part of the roller from within the eezStudio GUI. Only the main part is visible.

I was wondering if it would be appropriate to add this issue to the existing one on GitHub with the number #357, which was created earlier?
[image]

mvladic, 2024-05-09:
> Also I found that when a roller is defined it has a line which is not completed when it creates a screen in screen.c. I can solve it replacing LVOBJ_FLAG by LV_OBJ_FLAG_CLICKABLE

This is already reported and fixed in the master branch (see #337). The next version with this fix will be released by May 13.

mvladic, 2024-05-09:
> I've stumbled upon some new issues with the roller, this time within the eezStudio GUI. As you can see in the attached picture, it seems that there's currently no option available to control the style of the selection part of the roller from within the eezStudio GUI. Only the main part is visible.

Fixed.

closed 2024-05-10 by mvladic