eez-open/studio #360: Possible bugs in importing project process

glt248, 2024-05-07:
Hello,
When you import 2 instances of a same project :
[image]

We can see widgets for the two instances : 
[image]

But instances global variables are not distinct : 
[image]

So if I increment on variable, the two have the same value. I can't use two instances of the same project.

mvladic, 2024-05-07:
Can you explain the purpose of importing the same project twice?

glt248, 2024-05-07:
I'm using project to create a front end User Widgets and some User Actions in order to communicate with a DC/DC converter. So that is a under projet.

In a main project, I'm importing two instances of this controller because I'm using 2 DC/DC in a test bench...

glt248, 2024-05-07:
Ok, thank you for this reply, I will have a look on your proposal.

closed 2024-05-10 by mvladic