eez-open/studio #363: Circular display

LazaroFilm, 2024-05-11:
I would love to see an option for circular displays. I'm using [this module](https://www.waveshare.com/wiki/RP2040-Touch-LCD-1.28) with LVGL. To visualize the circular display I have to fidget with a temporary overlayed round shape as mask. It would be great to have an option to display a circular mask over the display to get a better and faster preview of the final screen.

mvladic, 2024-05-11:
Something like this?

[image]

LazaroFilm, 2024-05-12:
Yes!!! Exactly like this!!

mvladic, 2024-05-13:
Circular display option has been added in Settings:

[image]

prasimix, 2024-05-16:
Fixed

closed 2024-05-16 by prasimix