eez-open/studio #369: no screen to see with ArduinoIDE and JC8048W550_C ESP32S3 N16R8

hape65, 2024-05-15:
hi
i compile this Source with LVLG 8.x.x

```
/*******************************************************************************
 * Start of Arduino_GFX setting
 ******************************************************************************/
#include <Arduino_GFX_Library.h>

/* OPTION 1: Uncomment a dev device in Arduino_GFX_dev_device.h */
#include "Arduino_GFX_dev_device.h"

/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/

/*******************************************************************************
 * Start of EEZ Settings
 ******************************************************************************/

//#include ".h"
#include "touch.h"
#include "lvgl.h" 

#include "ui.h"

/*******************************************************************************
 * End of EEZ Settings
 ******************************************************************************/

#ifdef ESP32
#undef F
#define F(s) (s)
#endif

/* Change to your screen resolution */
static uint32_t screenWidth;
static uint32_t screenHeight;

void setup()
{
  Serial.begin(115200);
  // Serial.setDebugOutput(true);
  // while(!Serial);
  Serial.println("Arduino_GFX PDQgraphicstest example!");

#ifdef GFX_EXTRA_PRE_INIT
  GFX_EXTRA_PRE_INIT();
#endif

  // Init Display
  if (!gfx->begin())
  // if (!gfx->begin(80000000)) /* specify data bus speed */
  {
    Serial.println("gfx->begin() failed!");
  }                                                 // digit size

#ifdef GFX_BL
  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);
#endif
  gfx->fillScreen(RED);
  delay(500);
  gfx->fillScreen(GREEN);
  delay(500);
  gfx->fillScreen(BLUE);
  delay(500);
  gfx->fillScreen(BLACK);
  delay(50);
  lv_init();
  delay(10);
  
  screenWidth = gfx->width();
  screenHeight = gfx->height();

Serial.println("before touch");

touch_init();

Serial.println("before ui_init");

ui_init();

Serial.println("after ui_init");
}

static inline uint32_t micros_start() __attribute__((always_inline));
static inline uint32_t micros_start()
{
  uint8_t oms = millis();
  while ((uint8_t)millis() == oms)
    ;
  return micros();
}

void loop(void)
{  
  lv_timer_handler();
  ui_tick();
}
```

this is what i see in the Serial Monitor:

> 16:11:04.698 -> Rebooting...                                           
> 16:11:04.698 -> �ESP-ROM:esp32s3-20210327                              
> 16:11:04.698 -> Build:Mar 27 2021                                      
> 16:11:04.698 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
> 16:11:04.698 -> Saved PC:0x42046582                                    
> 16:11:04.698 -> SPIWP:0xee                                             
> 16:11:04.698 -> mode:DIO, clock div:1                                  
> 16:11:04.698 -> load:0x3fce3808,len:0x4bc                              
> 16:11:04.698 -> load:0x403c9700,len:0xbd8                              
> 16:11:04.698 -> load:0x403cc700,len:0x2a0c                             
> 16:11:04.698 -> entry 0x403c98d0                                       
> 16:11:04.834 -> Arduino_GFX PDQgraphicstest example!                   
> 16:11:06.550 -> before touch  
> 16:11:06.550 -> before ui_init
> 16:11:06.550 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 16:11:06.581 -> 
> 16:11:06.581 -> Core  1 register dump:
> 16:11:06.615 -> PC      : 0x42005f83  PS      : 0x00060e30  A0      : 0x820054f1  A1      : 0x3fcebb60  
> 16:11:06.615 -> A2      : 0x00000000  A3      : 0x3fced708  A4      : 0x00000004  A5      : 0x420459d4  
> 16:11:06.615 -> A6      : 0x00000043  A7      : 0x3fce9dc0  A8      : 0x820162ca  A9      : 0x3fcebb40  
> 16:11:06.615 -> A10     : 0x3fced6e4  A11     : 0x00000000  A12     : 0x3fced6d0  A13     : 0x3fced6c0  
> 16:11:06.615 -> A14     : 0x00000021  A15     : 0x00000003  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
> 16:11:06.648 -> EXCVADDR: 0x00000022  LBEG    : 0x42016f7c  LEND    : 0x42016f9a  LCOUNT  : 0x00000000  
> 16:11:06.648 -> 
> 16:11:06.581 -> Core  1 register dump:
> 16:11:06.615 -> PC      : 0x42005f83  PS      : 0x00060e30  A0      : 0x820054f1  A1      : 0x3fcebb60  
> 16:11:06.615 -> A2      : 0x00000000  A3      : 0x3fced708  A4      : 0x00000004  A5      : 0x420459d4  
> 16:11:06.615 -> A6      : 0x00000043  A7      : 0x3fce9dc0  A8      : 0x820162ca  A9      : 0x3fcebb40  
> 16:11:06.615 -> A10     : 0x3fced6e4  A11     : 0x00000000  A12     : 0x3fced6d0  A13     : 0x3fced6c0  
> 16:11:06.615 -> A14     : 0x00000021  A15     : 0x00000003  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
> 16:11:06.648 -> EXCVADDR: 0x00000022  LBEG    : 0x42016f7c  LEND    : 0x42016f9a  LCOUNT  : 0x00000000  
> 16:11:06.648 -> 
> 16:11:06.648 -> 
> 16:11:06.648 -> Backtrace: 0x42005f80:0x3fcebb60 0x420054ee:0x3fcebb80 0x42004ec7:0x3fcebba0 0x42001f6d:0x3fcebbc0 0x4200205b:0x3fcebbe0 0x420020a7:0x3fcebc00 0x42001e4c:0x3fcebc20 0x42022162:0x3fcebc50
> 16:11:06.648 -> 
> 16:11:06.648 -> 
> 16:11:06.648 -> 
> 16:11:06.648 -> 
> 16:11:06.648 -> ELF file SHA256: 3f4d77ded5c00e02
> 16:11:06.689 -> 
> 16:11:06.846 -> Rebooting...                       

On the scree the GFX flashes red-green-blue-black

The ui is only one screen no actions with a field and a keyboard toched to a textarea

What is wrong that there is such an error and no GUI?

hape

mvladic, 2024-05-16:
Judging by the serial output, we can conclude that a crash occurred when calling `ui_init()`. Now it is necessary to find out exactly where in `ui_init` the crash occurred. So may I recommended that you insert print to serial output inside the `ui_init` function and deeper until you reach the line in EEZ Studio generated code that causes the crash.

hape65, 2024-05-16:
if i add lines like this in ui.h
i get 
C:\Users\hape\Documents\Arduino\PDQgraphicstest2_short\ui.h:14:7: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
 Serial.println("before screens.h");
So it seems this is not possible - what can i do to get .ino sketch with EEZ Studio to run?

hape65, 2024-05-16:
i want to know where the program breaks down - so my idea was to make a serial.println before and after each command. So i type in the line between each command this serial.println

```
#ifndef EEZ_LVGL_UI_GUI_H
#define EEZ_LVGL_UI_GUI_H

#include <lvgl.h>

#include "screens.h"

#ifdef __cplusplus
extern "C" {
#endif

void ui_init();

void ui_tick();

void loadScreen(enum ScreensEnum screenId);

#ifdef __cplusplus
}
#endif

#endif // EEZ_LVGL_UI_GUI_H
```

thats my ui.h

mvladic, 2024-05-16:
I can see that you are beginner in C programming. You can't add `Serial.print` line anywhere, it must be inside function body. So, try to add it inside some function in `ui.c`, for example start with `ui_init`. And, later you can tray in create_sceen functions located in `screens.c`.

hape65, 2024-05-16:
ok this
```
void ui_init() {    
    Serial.println("before create_screens");
    create_screens();
    Serial.println("before loadScreen");
    loadScreen(SCREEN_ID_MAIN);
    Serial.println("after loadScreen");
}
```
gives this:
C:\Users\hape\Documents\Arduino\PDQgraphicstest2_short\ui.c: In function 'ui_init':
C:\Users\hape\Documents\Arduino\PDQgraphicstest2_short\ui.c:29:5: error: 'Serial' undeclared (first use in this function)
     Serial.println("before create_screens");
     ^~~~~~

mvladic, 2024-05-16:
Add `#include "Arduino.h"` at the top.

hape65, 2024-05-16:
no that also doesn't work it should be something like this (but for now that also doesn't work
https://stackoverflow.com/questions/66632376/how-to-call-serial-print-from-c-file-in-arduino-ide

mvladic, 2024-05-16:
Stackoverflow answer is correct.

In ui.c add line `void my_log(const char *msg);` after includes and replace `Serial.println` with `my_log`, like this:

```
#include ...

// ...

void my_log(const char *msg); // new!

// ...

void ui_init() {    
    my_log("before create_screens"); // use my_log instead of Serial.println
    create_screens();
    my_log("before loadScreen"); // same here
    loadScreen(SCREEN_ID_MAIN);
    my_log("after loadScreen"); // same here
}
```

In your `.ino` file add at the end:

```
extern "C" void my_log(const char *msg) {
  Serial.println(msg);
}
```

I hope I didn't make some syntax error while writing this without trying. If you still have an error please post here output of the build.

mvladic, 2024-05-16:
Let me explain. Of course it `Serial.println` didn't work in `ui.c` file because `ui.c` is writen in C and Serial is C++ object with the `println` method/function. Also, extern "C" is required because of [C++ name mangling](https://www.geeksforgeeks.org/extern-c-in-c/)

mvladic, 2024-05-16:
And by the way, `.ino` file is actually `.cpp` file (C++ source file) but, I think, Arduino decided to invent a new extension so when you double click `.ino` file OS will open Arduino IDE.

mvladic, 2024-05-16:
One more thing. You can meet us at [Discord](https://discord.gg/dhYMnCB) where there is an ongoing discussion with other members and supporters about usage, improvements, issues, etc. For questions about EEZ Studio and LVGL support use following channels: #eez-studio, #lvgl and #eez-flow

closed 2024-05-17 by mvladic