eez-open/studio #374: [LVGL] "Unused bitmap" is reported although image exists on main page

RE2012, 2024-05-20:
**Describe the bug**
If you check or build a project then "Unused bitmap" is reported for every image although image exists on page.

**To Reproduce**
Steps to reproduce the behavior:
1. in studio go to "Home" page, select "examples" and select project "eez_lvgl_demo"
2. click "edit project"
3. click on "check" button
4. in output, see "unused bitmap: logo_eez" and "unused bitmap: logo_lvgl" 

**Expected behavior**
It should only report unused bitmaps.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 10

**LVGL version (if used)**
- 8.3, same with 9.0

**eez-studio version**
- 0.13.0

mvladic, 2024-05-20:
You are right, these are false warnings. It will be fixed. Thanks.

mvladic, 2024-06-12:
Fixed.

closed 2026-02-08 by mvladic