eez-open/studio #377: [LVGL] If flow support is turned off then the generated files still contain flow usages

RE2012, 2024-05-20:
**Describe the bug**
 If flow support is turned off then the generated files still contain flow usages.

**To Reproduce**
1. in studio go to "Home" page, select "examples" and select project "eez_lvgl_demo"
2. click "edit project"
3. click on "build" button
4. go to project settings and switch off "flow support"
5. click on "build" button
6. check the output folder and see that many files contain flow data, e.g.:
- flow_def.c and flow_def.h exist
- ui.c uses eez_flow_init and eez_flow_tick
- ui.h: #include <eez/flow/lvgl_api.h>

**Expected behavior**
If flow support is removed from a project (unchecked) nad project is build again then build files should not contain flow data in any code anymore.

**Desktop (please complete the following information)**
 - OS: Windows 10

**LVGL version (if used)**
- 8.3

**Additional context**
- Studio 0.13.0
- Arduino IDE 2.3.2
- Espressif 2.0.11 for ESP32 S3
- Background reason: additional flow support caused too many compilation problems so I wanted to get rid off it. (eez-framework is in libraries folder like this libraries\eez-framework )

mvladic, 2024-05-20:
When you turn off flow support, the template files, for now, are not changed/updated. "LVGL" and "LVGL with EEZ Flow" have different set of template files. This is already reported (see #366) and will be addressed in the next release. For now, you will need to manually delete all the template files and copy-paste from a new project created with the "LVGL" template.

RE2012, 2024-05-27:
Thank you. I have tried it with 0.13.1 and copied all files from sample LVGL no flow project. However in screens.c I still see getFlowState usages. For example:
void create_screen_main() {
...
            lv_obj_set_local_style_prop(obj, LV_STYLE_BG_OPA, value, LV_PART_MAIN);
            lv_obj_set_local_style_prop(obj, LV_STYLE_BORDER_WIDTH, value, LV_PART_MAIN);
            create_user_widget_header(obj, **getFlowState**(flowState, 0), 33);

and:
void tick_screen_main() {
    tick_user_widget_header(**getFlowState**(flowState, 0), 33);

Looking for next version.
Best Regards

mvladic, 2024-05-27:
> Thank you. I have tried it with 0.13.1 and copied all files from sample LVGL no flow project. However in screens.c I still see getFlowState usages.

This is already fixed in master branch, see #373. Will be release in 0.14.0, due by June 10, 2024, but you can always build a Studio from master branch.

tylercamp, 2024-06-02:
A similar problem happens in the reverse case - when I enable Flow in a project which did not originally have it, the relevant files + content updates don't seem to be applied. Tried latest release and latest on `master`.

Would it be appropriate to copy/paste the relevant JSON from an LVGL+Flow project file into my originally-LVGL-only project file as a workaround?

mvladic, 2024-06-02:
Yes, you can copy and paste, but you don't need to go into JSON, you can do it inside Studio. 

[image]

[image]

[image]

closed 2024-09-26 by mvladic