eez-open/studio #379: [Dashboard] New widget for tabular display of data

mvladic, 2024-05-21:
Add new widget for tabular display of data from the database or imported files.

Table cells should be able to contain:

- simple label
- tree label
- progress
- checkbox or 
- image. 

Enable sorting by columns, pagination, search and export to HTML, PDF (includes #12) and CSV.

fietser28, 2024-05-31:
I would like to have  a button to 'load column from data': If I specify a structure type variable in the data field pressing this button will disable auto column and defines all the columns based on the elements of the structure pointed to in the data field. This allows to quickly make an editor/display for existing data structures.

Manually performing this action is very cumbersome, also because I don't know how to display both the definition of the structure and the definition of the table column at the same time. So for each element I have to switch in UI.

fietser28, 2024-05-31:
The auto column mode doesn't resolve enumerations. It shows numbers.

fietser28, 2024-05-31:
Using a array of struct (in screenshots: profiles is array of  batteryProfile struct) as data input: If the struct contains a double the value displayed is incorrect, both when using plaintext and money (which appears to the number formatting type):

[image]
[image]
[image]
[image]

I think this is a very important use case.

mvladic, 2024-05-31:
> The auto column mode doesn't resolve enumerations. It shows numbers.

Auto column is implemented by the Tabulator library. But, it makes sense if input data comes from `array:struct:...` to override tabulator implementation since in this case Studio knows better.

fietser28, 2024-05-31:
> > The auto column mode doesn't resolve enumerations. It shows numbers.
> 
> Auto column is implemented by the Tabulator library. But, it makes sense if input data comes from `array:struct:...` to override tabulator implementation since in this case Studio knows better.

I agree but think the user knows even better but it is a lot of work to manually define all the columns that represent the struct elements. If there is a button that generates the column definitions with some sensible defaults, the users can easily customize the column afterwards. Pressing the button is a 1 time action.

mvladic, 2024-05-31:
Sure, both features should be added.

fietser28, 2024-05-31:
> Using a array of struct (in screenshots: profiles is array of batteryProfile struct) as data input: If the struct contains a double the value displayed is incorrect, both when using plaintext and money (which appears to the number formatting type):

Juist noticed this problem is also present in the JSON Stringify action. So probably this is a more generic issue regarding internal struct -> json conversion.

fietser28, 2024-06-01:
> Juist noticed this problem is also present in the JSON Stringify action. So probably this is a more generic issue regarding internal struct -> json conversion:

Did some more test. It appears only Double type is not working, other variable types work fine:
[image]

[image]

mvladic, 2024-06-01:
I found what is the problem with doubles in struct. It only appears when struct is defined through "Default value". If you make a struct with `Flow.makeValue("struct:mystruct", { ... })` it will work fine. The root cause for the problem is that doubles are not 8 byte aligned. I will commit the fix soon.

mvladic, 2024-06-01:
Fix for doubles commited.

fietser28, 2024-06-01:
It works for all my use cases.

mvladic, 2024-06-01:
I added command "Build From Structure Definition":

[image]

Also, when "Auto columns" is enabled we will override auto columns feature from Tabulator in case of table data is of structure type and infer columns info from Structure definition.

For now, only Title and Field is inferred. In the future we will try to infer other Column properties, please be free to suggest it.

mvladic, 2024-06-01:
I forgot enum types, we should build formatter property from that.

fietser28, 2024-06-01:
This is how I'm now manually configured enum (still working out the exact settings for my case):
[image]

I think the generation/sync of the parameter list is important/convenient.

mvladic, 2024-06-01:
It doesn't sync for now. For sync we should have a new property:

Auto columns         [ ]
Sync columns with structure definition: [x] ____my_struct [V]

mvladic, 2024-06-01:
This is required because if you build from structure and changed some column property you don't want sync to overwrite it. So you are all for sync or no sync.

fietser28, 2024-06-01:
Working on my example I'm hitting a limitation when working with selections. Selections are nice to allow for the deletion, duplication, etc of rows in my use case:

[image]

Please add 'row selected' and 'row deselected' events to the widget. The event should return the index number of the index of the row selected (not the displayed order but the index of the data array given to the tablulator).
See [https://tabulator.info/docs/6.2/events#select](https://tabulator.info/docs/6.2/events#select) 

These events will allow me to implement a 'delete row' , 'duplicate row' and a 'use row' functions.

mvladic, 2024-06-01:
I have added ON_ROW_SELECTED and ON_ROW_DESELECTED events. Both events outputs json value:

```
{
  index: number, // 0-based index in table data
  position: number // 1-based position in displayed rows
}
```

You will use `.index`.

mvladic, 2024-06-01:
Also, columns properties for enum types are also generated:

[image]

But, this is still work in progress.

prasimix, 2024-06-07:
Fixed

closed 2024-06-07 by prasimix