eez-open/studio #380: Pages not visible, but widgets exist - export works fine.

anujdeshpande, 2024-05-21:
**Describe the bug**

I have a few pages, and some of them don't show the widgets that I have on them. This issue started in the middle of the project - so I can verify that the export still has them. Plus I can see the widgets still listed when I select the said pages.
Note that this doesn't happen with all the pages, only some.

**To Reproduce**
Steps to reproduce the behavior:
I don't know how to reproduce this.

**Expected behavior**
I should be able to see the widgets.

**Screenshots**
Page S2_1 should be visible just as S2

[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11

** LVGL version (if used)**

- 8.3

**Additional context**
Add any other context about the problem here.

mvladic, 2024-05-21:
Please attach your eez-project file.

anujdeshpande, 2024-05-21:
[Uploading EEZproj.zip…]()
@mvladic attached here

anujdeshpande, 2024-05-21:
[EEZproj.zip](https://github.com/eez-open/studio/files/15393948/EEZproj.zip)
@mvladic attached here ^

mvladic, 2024-05-21:
This is the reason:

[image]

I will fix this. In the meantime change "Variable" to "Literal".

mvladic, 2024-05-21:
Fixed.

anujdeshpande, 2024-05-21:
@mvladic Thanks! That seems to have fixed the visibility issue. But that breaks the creation of the `get_` functions that were being created. I see that there are event callbacks now - is that a replacement to those functions? 

Having a blank text field for a variable Label doesn't seem to create an event handler. That works for a text field though.

mvladic, 2024-05-21:
> But that breaks the creation of the get_ functions that were being created.

Hm, sorry but I don't understand what you mean. If you mean `get_` functions for the variables, in your project there is no variable defined. 

Please check the following example:

[image]

Here you can see how variables can be used to show dynamic content inside Label widget and also how to use user defined actions that can be attached to the event handler.

mvladic, 2024-05-21:
Also, please be aware that if you are using variables and user actions that are implemented in C/C++ then you can't test it inside Studio (Run button). But. if you create "LVGL with EEZ Flow" project then you can use EEZ Flow (type of visual programming) to add interactivity to your project. This is great for quick prototyping.

closed 2024-05-23 by anujdeshpande