eez-open/studio #383: [LVGL] Allow Arc `Range min`  and `Range max` to be assignable

scruplelesswizard, 2024-05-26:
**Is your feature request related to a problem? Please describe.**

I have several arc widgets that are embedded in user widgets for reuse. However, the range for those arc widgets is dynamic based on other variables (i.e. two servos are configured, one with a 180 degree arc, and the other with a 270 degree arc. Each instance of the user widget controls one of the servos.)

**Describe the solution you'd like**

I want to be able to add an input to the arc widget and assign that input to the value of `Range min` and `Range max`

**Describe alternatives you've considered**

- Reconfigure the widget dynamically at runtime

**Additional context**

- None

mvladic, 2024-05-27:
Just to clarify, "Assignable" is used when the property has read/write semantics, such as the "Value" property in the Arc widget, because the Widget can also set a "Value". In the case of "Range min" and "Range max", when the widget never changes them, it will be Expression (if using EEZ Flow) or Variable (if not using EEZ Flow).

I will add this in the next version.

scruplelesswizard, 2024-05-27:
Expression or variable would be fantastic! Thank you ❤️

mvladic, 2024-06-07:
Done.

closed 2024-06-07 by prasimix

scruplelesswizard, 2024-09-22:
Thank you! This is amazing