eez-open/studio #391: [Dashboard] automatic json to array of struct conversion results in instability

fietser28, 2024-06-01:
**Describe the bug**
Using the tabulator with an array of struct: When I update the array if something is changed/editted in tabulator everything seems to work fine, but after that a Array.append( array, ...) throws an error the arrays is not an array.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit a value in the table, press enter to store
2. Select a row
3. Press 'duplicate row'
4. See error

**Expected behavior**
Duplicate keeps working after an edit. 
If Step 1 is omitted everything works fine. I suspect the action in step 1 causes some kind of corruption due to the conversion of json to array of struct.

**Screenshots**
Step 1 causes the following action to execute:
[image]

Step 2 executes:
[image]

Step 3 executes:
[image]

Error:
[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: 14.2.1

fietser28, 2024-06-01:
[BatteryDischarge.eez-dashboard.zip](https://github.com/user-attachments/files/15522777/BatteryDischarge.eez-dashboard.zip)

The dashboard to test with.

mvladic, 2024-06-01:
> The dashboard to test with. 

There is no eez-project in the zip file.

fietser28, 2024-06-01:
[BatteryDischarge.eez-project.zip](https://github.com/user-attachments/files/15522897/BatteryDischarge.eez-project.zip)
Retry...

fietser28, 2024-06-01:
Also noticed the same error happens after loading a file. You can easily create a json by saving the default values to a file first.

mvladic, 2024-06-02:
Fixed.

closed 2024-06-02 by fietser28