eez-open/studio #394: UI issues with recent commits

fietser28, 2024-06-02:
**Describe the bug**
Commit [https://github.com/eez-open/studio/commit/26edf14356bdf3a423f0e5640ef0b6275f0fee98](https://github.com/eez-open/studio/commit/26edf14356bdf3a423f0e5640ef0b6275f0fee98) introduced some strange UI behaviours: 
- adding a new action to a dashboard project by dropping it on the screen canvas: the alignment lines stay present after releasing the action. The object on the canvas appears frozen.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a dashboard project
2. drag a new action or widget on the canvas
3. Canvas is 'frozen' / scrambled 

Other case:
1. Open a LVGL project
2. drag a new action on the canvas
3. new action is displayed malformed and frozen.

**Expected behavior**
As before commit.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 14.2.1

**Additional context**
Reverting back before this commit everything works.

mvladic, 2024-06-02:
Fixed!

fietser28, 2024-06-02:
In dashboards it seems fixed. In LVGL project I still see strange behaviour:
[image]

These are newly created actions. Existing actions/widget seem fine.

mvladic, 2024-06-02:
Fixed.

fietser28, 2024-06-02:
Just dit a duplicate of a label and this showed up:
[image]

When moving the new label inside the screen in the widget structure the label contents stil won't show:
[image]

After removing the ' weird displayed actions' in the comment above, all widgets suddenly started working:
[image]

This is consistent if I reproduce this: Adding a new ACTION the action shows up weird and widget start behaving weird. After deletion of the new action everything is normal again.

**I was typing this while you fixed this, so will test first..**

fietser28, 2024-06-02:
Dit tests. Most use cases from my last comments are fixed now.

Only 1 stil remains: When I duplicate a widget (right click with mouse on original and select duplicate) it shows up wrong:
[image]

Dragging it into the screen in widget structure fixes it again.

mvladic, 2024-06-02:
Fixed.

fietser28, 2024-06-02:
All fine now!

closed 2024-06-02 by fietser28