eez-open/studio #399: [Flow] UI improvement on editable outputs that shouldn't be deleted and consistency between this.

fietser28, 2024-06-05:
**Is your feature request related to a problem? Please describe.**
This is just a consistency/UI improvement:
The UI currently allows you to delete or rename certain outputs of actions that should not be renamed or deleted. Deleting or renaming will trigger an error during execution and is never a valid use case (imho). Examples actions are: Eval JS, CSVParse and JSONParse. I'm also not sure if this is currently consistent between all actions (Eval and Watch don't allow you to edit/delete the output type).

**Describe the solution you'd like**
Disable the rename and delete buttons for these outputs.
All actions should have consistent implementation.

**Describe alternatives you've considered**
Hide these type of  outputs in the custom outputs list, but in that case the output type can't be set by the users.
Alternative would be to add a fixed UI item to choose the output type under the specific group of the action.

mvladic, 2024-06-05:
These action allows output editing so you can change output type, from example by default JSONParse will output 'json' but you can change it to your structure type. But, I agree we should do something about this problem when you change the name or completely delete it - we could at least report this as an error in Checks tab.

fietser28, 2024-06-05:
Error in the Checks tab is a good idea.

How about the alternative: add a fixed UI item to choose the output type under the specific group of the action?

I'm also wondering if the 'Eval' result output should have a user definable type?