eez-open/studio #400: Add search field to expression builder

fietser28, 2024-06-05:
**Is your feature request related to a problem? Please describe.**
No, just UI enhancement

**Describe the solution you'd like**
In the expression builder I would like to have a search field. This search field filters the options shown by the expression builder. This is nice because sometimes the list in the expression builder are long, see screenshot below. The filter in the variable tab shown in the background works like a charm, so would also like to have this in the expression builder.

[image]

fietser28, 2024-06-05:
Small addition: Maybe also add a sync option to the search field just like in the SCPI terminal: If you start typing in the edit field a search is automatically performed.

mvladic, 2024-06-18:
Implemented.

[image]

> Small addition: Maybe also add a sync option to the search field just like in the SCPI terminal: If you start typing in the edit field a search is automatically performed.

I haven't done this yet. It makes more sense for SCPI commands, since there is a help page associated with the command. Until we can do the same for the expression parts, I think there is no point to add sync option.

fietser28, 2024-06-18:
> I haven't done this yet. It makes more sense for SCPI commands, since there is a help page associated with the command. Until we can do the same for the expression parts, I think there is no point [...]

I agree.

It is working fine. Thank you.

closed 2024-06-18 by fietser28