eez-open/studio #403: [LVGL] Adjust Arc widget knob size separately from the arc width

APF-MRoger, 2024-06-07:
Unless I am a bit daft (it happens), it doesn't seem to be possible to adjust the Arc knob size for LVGL projects, separately from the Arc width. The offset of size between the Arc width and the knob seems to be always the same.
It is possible to adjust it in the screens.c file but that takes away the simplicity of using EEZ studio.

**Describe the solution you'd like**
A field in the Arc properties to adjust the size of the knob in pixels.

**Describe alternatives you've considered**
As mentioned, it is possible to add a line for each knob in screens.c to adjust the size of the knob, or to do it in Setup. It breaks the seamless EEZ Studio experience

**Additional context**
N/A

mvladic, 2024-06-07:
How would you set knob size using LVGL API?

martinroger, 2024-06-07:
> How would you set knob size using LVGL API?

Hey, replying from my non-work account. So there isn't a direct way to set the size via the API as far as I am aware off, however setting Padding Top/Left/Right/Bottom to negative pixel values can achieve the desired effect. 
Sorry I forgot to mention that this is more a hack than a direct API usage (which EEZ studio is doing very well ) !

mvladic, 2024-06-07:
Good tip! But, actually, setting positive value for padding in Knob part of Arc can be used to increase knob size:

[image]

If you want to remove knob then you can use negative padding values.

martinroger, 2024-06-07:
Right right. I guess the issue can be closed then ? This sounds like a good enough work-around to me...

closed 2024-06-07 by mvladic