eez-open/studio #405: [question] What is Flow?

paaspaas00, 2024-06-07:
In the exported C files I see flow_def.c, flow_def.h, and there are multiple refs to it in the other files.
They're a bit obscure to me, what is their purpose? Is it an eez lib? Is it going to be compiled when targeting Arduino too? Is it big? Can it be dropped safely in case?

Thank you since now!

mvladic, 2024-06-07:
This file is only required if you want to use EEZ Flow: https://github.com/eez-open/studio/wiki/Q&A#what-is-difference-between-lvgl-and-lvgl-with-eez-flow-project-templates

mvladic, 2024-06-07:
> what is their purpose? 

It contains flow binary definition, something like byte code.

> Is it an eez lib? 

No, eez-framework lib is required to interpret this flow binary definition.

> Is it going to be compiled when targeting Arduino too? 

Yes, if you are opted in for using EEZ Flow.

> Is it big? 

Depends on the size of your project.

> Can it be dropped safely in case?

No, if you are using EEZ Flow.

paaspaas00, 2024-06-07:
@mvladic Cool! Thank you for the nice explanation!

closed 2024-06-07 by mvladic