eez-open/studio #406: Add warning to check for unused variables, user_actions and user_widgets

fietser28, 2024-06-08:
**Is your feature request related to a problem? Please describe.**
Not a problem, just convenience to allow for easier cleanup of projects.

**Describe the solution you'd like**
The automatic checks give warnings regarding to unused variables, structs, enums, user_actions and user_widgets.

**Describe alternatives you've considered**
Add a button per panel to search for these and or delete.

mvladic, 2024-06-08:
Makes sense. However, it could be slow to do it in the background like, for example, when we check if expressions are correct. We can do these additional checks when you press the Build button.

mvladic, 2024-06-08:
Also, we can report warning if some property of widget or action is of specific type and when analyzing given expression we conclude that this expression gives value of another type. So these warning will report something: "type mismatch, property X expects type integer but expression evaluates to boolean". This doesn't mean that your flow will result in error, because we can implicitly convert `boolean` to `integer`, but it could be sign that something is not entirely correct. This type of checks we can do in the background. 

We can add Warning level setting. warning level 1 is what we have now, warning level 0 could be no warnings, warning level 2 we can do this "type mismatch" checking, for example.