eez-open/studio #411: Small UI tweaks

DrKropp, 2024-06-12:
Please allow me to offer some suggestions for some tweaks to improve the workflow within Studio:

1. Automatically toggle the checkbox for a property when clicking on the box next to that property (text entry or dropdown). I keep hitting the box, typing a value, noticing that the field isn't active, hitting the checkmark, then doing it again.
2. Allow renaming of objects (pages, variables, etc) from the right click menu on the item; next to the add/delete buttons on the tab header, or by double clicking the item name instead of using the properties tab
3. Add an option to select multiple font sizes when adding a font instead of opening the font repeatedly to choose all the desired sizes sequentially
4. Add a "Style Builder" wizard that allows basic customization (background, background image, text color, fonts, border styles, etc.) for the most commonly adjusted parameters across **ALL** states in that project. Manually tweaking each state individually affords too many opportunities to miss a field or input a value incorrectly. Reserve the properties tab to modify each state afterwards.
5. Allow placeholder or dummy text for labels that contain an expression to make it easier to layout elements accurately instead of showing the expression
6. Include the ability to use formatting options for numeric values in labels that contain an expression. My labels dance around when going from e.g. 1 to 1.1, so it would be nice to apply formatting and display 1.0 going to 1.1.

Please don't take these as criticism or that they have to be addressed if it is too laborious. This is a fantastic project, these are just some  things I have noted while developing an LVGL project.

mvladic, 2024-06-12:
First of all, thank you for your suggestions and comments!

I think points 1 to 3 are all good ideas. Here are my comments regarding remaining points:

> 4. Add a "Style Builder" wizard ...

It may not be exactly the same as what you suggest, but we already have support for reusable style definitions in Styles tab (you can find it on the top-right side):

[image]

[image]

[image]

You can select style for the widget in "Use style" property:

[image]

You can even set a default stye for the widget type:

[image]

What we are missing is a way to display only most frequently used style properties for the widget type. 

> 5. Allow placeholder or dummy text for labels that contain an expression to make it easier to layout elements accurately instead of showing the expression

This is already implemented in the latest version 0.14.0:

[image]

> 6. Include the ability to use formatting options for numeric values in labels that contain an expression. My labels dance around when going from e.g. 1 to 1.1, so it would be nice to apply formatting and display 1.0 going to 1.1.

There is already a function `Math.round(/*value*/, /*numOfDigits*/)`. But, this is not enough. We need more options, for example to always display trailing zeros (1.0 instead of 1) etc.

mvladic, 2024-06-14:
In version 0.14.1, point "1. Automatically toggle the checkbox for a property ..." is implemented.

LazaroFilm, 2024-06-27:
Four point number five, it would be great to have the ability to create a slider that can change the preview value of the variable to see how the design behaves when that value is changed without having to constantly type different numbers in the preview value box.