eez-open/studio #415: [dashboard] Uncaught RuntimeError: memory access out of bounds

dong-king, 2024-06-14:
**Describe the bug**
When I tested a certain action separately, it was normal. But when I put it into the project, it sometimes exposed a memory error：memory access out of bounds.
I don't know if it is because my dashboard project has many business functions, which caused the memory to exceed the bounds.

Is the memory size defined in the memory.h file of eez-framework:
```
extern uint8_t g_memory[];
static uint8_t * const MEMORY_BEGIN = g_memory;
static const uint32_t MEMORY_SIZE = 64 * 1024 * 1024;
```
Is there any way to dynamically adjust the memory size?

**Screenshots**
[image]

mvladic, 2024-06-14:
I don't think this is memory size issue. If you can post the eez-project file with which I can reproduce this problem it would be of great help.

dong-king, 2024-06-14:
Sometimes I want to execute a slightly complex js code directly in the action, so I customize an action based on your EvalJSExprActionComponent, as follows:
```
export class JsActionComponent extends ActionComponent {
    static classInfo = makeDerivedClassInfo(ActionComponent.classInfo, {
        componentPaletteGroupName: "Dashboard Specific",
        label: () => "JS",
        componentPaletteLabel: "JS",
        properties: [
            {
                name: "expression",
                type: PropertyType.MultilineText,
                propertyGridGroup: specificGroup,
                monospaceFont: true,
                disableSpellcheck: true,
                flowProperty: "template-literal"
            },
            makeExpressionProperty(
                {
                    // The escape character corresponding to '{'
                    name: "leftEscape",
                    type: PropertyType.MultilineText,
                    propertyGridGroup: specificGroup
                },
                "string"
            ),
            makeExpressionProperty(
                {
                    // // The escape character corresponding to '}'
                    name: "rightEscape",
                    type: PropertyType.MultilineText,
                    propertyGridGroup: specificGroup
                },
                "string"
            )
        ],
        beforeLoadHook: (
            component: JsActionComponent,
            jsComponent: Partial<JsActionComponent>
        ) => {
            if (
                !jsComponent.customOutputs ||
                jsComponent.customOutputs.length == 0
            ) {
                jsComponent.customOutputs = [
                    {
                        name: "result",
                        type: "any"
                    }
                ] as any;
            }
        },
        check: (component: EvalJSExprActionComponent, messages: IMessage[]) => {
            const { valueExpressions } = component.expandExpressionForBuild();

            valueExpressions.forEach(valueExpression => {
                try {
                    checkExpression(component, valueExpression);
                } catch (err) {
                    messages.push(
                        new Message(
                            MessageType.ERROR,
                            `Invalid expression "${valueExpression}": ${err}`,
                            getChildOfObject(component, "expression")
                        )
                    );
                }
            });
        },
        icon: (
            <svg viewBox="0 0 1664 1792">
[... 170 lines cut ...]
                assets.projectStore.outputSectionsStore.write(
                    Section.OUTPUT,
                    MessageType.ERROR,
                    err,
                    getChildOfObject(this, "expression")
                );

                dataBuffer.writeUint16NonAligned(makeEndInstruction());
            }
        });
    }
}

registerClass("JsActionComponent", JsActionComponent);
```
Because `{` and `}` cannot be used, I provide `leftEscape` (default is `#l#`) and `rightEscape` (default is `#r#`) to escape `{` and `}`.

The following screenshot is a simple usage:
[image]
This custom action works fine in most cases. But sometimes it prompts: Uncaught RuntimeError: memory access out of bounds and cannot read properties of null (reading 'context').
[image]

I cant find the reason.

BTW If you think this action is useful, you can merge it into eez-studio.

mvladic, 2024-06-14:
> BTW If you think this action is useful, you can merge it into eez-studio.

Yes, this is useful. I was aware of this issue with `{` and `}`, but I wasn't sure which way is the best to resolve it. I think maybe better option is to set left escape and right escape for the EEZ Flow expressions not for the JavaScript's `{` and `}`.

mvladic, 2024-06-14:
BTW There is `JSON.parse` action, so you don't need to use EvalJS for that.

mvladic, 2024-06-14:
I tried your Action with the same JavaScript code and I didn't see the problem so far. Can you tell me here:

[image]

What type you selected for the `result` output and `in` input?

mvladic, 2024-06-14:
This particular error I had before:

[image]

and it is fixed now in the latest version of Studio. You are probably on some older version.

dong-king, 2024-06-14:
> What type you selected for the `result` output and `in` input?
`result` and `in` correspond to my custom structure. 
The previous screenshot is just to illustrate how to use this action, not the action that will cause an error.

In fact, an action that cause errors will not fail every time. 
I have a dashboard project with a lot of business logic. When I put the actions that cause errors in this project into a new dashboard project, they will not cause errors.

Maybe there are other reasons in this complex dashboard project that cause errors.

Basically, the errors occur when `const factoryFn = new Function(factoryFnCode);`  or `await fn(
            ...moduleNames.map(moduleName => globalModules[moduleName])
        );`is executed.

mvladic, 2024-06-14:
Can you log the value of `inputSelectInfoStr`, I want to know the value when error occurs. Also, how structure WAVE_TABLE_SOURCE_... is defined?

dong-king, 2024-06-14:
> This particular error I had before: [...]
My colleague synchronized to version 0.13.0.

mvladic, 2024-06-14:
Never mind, this error in lz4.js isn't important.