eez-open/studio #416: [Dashboard] Enhance blob support

fietser28, 2024-06-14:
**Is your feature request related to a problem? Please describe.**
No just a feature request.

**Describe the solution you'd like**
Add a few functions to better support handling and processing of blob contents:

- [ ]  .toString() converts a blob to a string
- [ ]  .size() returns the size of the blob content in bytes. This is important for pre allocating structures when extracting data from blob
- [ ]  .slice(/*blob*/, /*start*/, /*[end]*/). This is important to avoid converting unnecessary amounts of data.

**Describe alternatives you've considered**
Alternative for .toString() is: 
[image]
I don't have alternatives for the other methods (probably can be done in JS Eval)

Feel free to complement/change.

mvladic, 2024-06-14:
`Array.length(blob)` works. But, we can also add `Blob.size()`.

FYI These operations also currently exists for the blob:

- addition (+): concatenation of two blobs
- equal (==): comparation of two blobs

mvladic, 2024-06-14:
Also, you can change blob content, i.e. `blob[i]` is allowed for the "Variable" property in SetVariable action.

fietser28, 2024-06-14:
> `Array.length(blob)` works. But, we can also add `Blob.size()`.

Maybe a good idea: To me it was not clear this was possible.

mvladic, 2024-06-14:
> .slice(/blob/, /start/, /[end]/). This is important to avoid converting unnecessary amounts of data.

What you mean by "This is important to avoid converting unnecessary amounts of data"?

fietser28, 2024-06-14:
> > .slice(/blob/, /start/, /[end]/). This is important to avoid converting unnecessary amounts of data.
> 
> What you mean by "This is important to avoid converting unnecessary amounts of data"?

If I only need to convert a small amount of the blob to a string:
[image]
In this case I only need the first 16 bytes of the blob in a string. The blob can be >1Mbytes. So it save converting and copying  a lot of data in this case.