eez-open/studio #417: [Dashboard] Large Array.alloc()  stops flow.

fietser28, 2024-06-15:
**Describe the bug**
When using Array.alloc for larger values (>1 M) the flow freezes on the alloc.

**To Reproduce**
Steps to reproduce the behavior:
[image]
Array is an array of integer.
NOOP will never run.

**Expected behavior**
The alloc works or an error is generated if the allocation is to large.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 14.2.1

mvladic, 2024-06-15:
This is a bug. However, when I fix this bug it will still be really slow when you are inside debug mode because of such a big array. This can be improved, but it is not trivial. So, we will keep this issue open until I improve debug mode performance.

Here are some debug mode implementation details. Debug mode is first implemented so that it works over serial/ethernet for the purpose of debugging applets on BB3. So, every variable change is transferred from the BB3 to the PC over serial/ethernet. We kept this even for the Dashboard projects although it can be done better since there is a shared memory between Studio and Dashboard runtime, so we don't need to transfer variables like it is serial/ethernet. In your case when you create array of 1000000 elements, the transfer of this array from runtime to debugger is really slow.

mvladic, 2024-06-15:
If you just Run your dashboard it should be really fast since there is no variable transfer that I described. However, at one point if you continue to increase array size you will hit memory limits.

mvladic, 2024-06-15:
Fixed.

fietser28, 2024-06-16:
Did some tests. With an alloc of 140k doubles the delay is not relay bad on my machine in debug.

If I switch to >1milion the first allocation seems to work, but the second fails. If it fails there is no error generated. I think that should be the case.

mvladic, 2024-06-16:
> If I switch to >1milion the first allocation seems to work, but the second fails. If it fails there is no error generated. I think that should be the case.

Second allocation?

fietser28, 2024-06-16:
[image]

With dataLength2 = 1750000. The first alloc (for x) appears to be OK, the second (for y) fails. See the watch panel:
[image]

 A few steps later the flow will fail when there is a try to write to y (a write to x works).

mvladic, 2024-06-16:
Try the latest version.

fietser28, 2024-06-16:
Something is still not working, but I can't reliably reproduce it.

mvladic, 2024-06-16:
It seams that some memory is retained more time than necessary, In your case where you are allocating a significant amount of memory this can be problematic. It is fixed now, so try with the latest version.

fietser28, 2024-06-16:
Found my own inconsistency.
Have a look at the attached flow (with debug): I would expect an error if an allocation fails, but it just continues.
[alloctest2.eez-project.zip](https://github.com/user-attachments/files/15858827/alloctest2.eez-project.zip)

It seems the allocation per object is limited not the total amount of memory. Is there a way to query the maximum allocation size and to have a flow act on it (downsample/slice/...)?

mvladic, 2024-06-16:
> I would expect an error if an allocation fails, but it just continues.

You are right, it should throw and error, not just return null value.

> It seems the allocation per object is limited not the total amount of memory. 
> Is there a way to query the maximum allocation size and to have a flow act on it?

Total memory available for dynamic allocations is limited to this number:

[image]

This can be increased. I need to investigate what is the best option to handle memory for dashboard projects. Do you have any show stopper right now, can you continue with your project?

mvladic, 2024-06-16:
Our own memory allocator is currently used in Dashboard projects. The reason is that at one point I wanted to test it on real world cases and later I didn't change it. We allocate 64 MB at the beginning of flow execution and this allocator works with this memory. We can change that and use emscripten default allocator in which case much more memory will be available. Only, I don't know how to query how much memory is used.

fietser28, 2024-06-16:
> This can be increased. I need to investigate what is the best option to handle memory for dashboard projects. Do you have any show stopper right now, can you continue with your project?
> 

It is not a show stopper. I can continue. Thanks for asking.

mvladic, 2024-06-16:
I switched to emscripten memory allocator for dashboards. Much more memory is now available. Please note, when you run project alloctest2.eez-project in Debug mode it will crash studio at one point because it will run out of memory (debug mode uses much more memory for its own purpose), but in Run Mode it will work (it will also eventually run out of memory available to emscripten, i.e. web assembly).

mvladic, 2024-06-16:
I would really like to improve performance of the Debug mode. Therefore, let this issue remain open.

mvladic, 2024-06-16:
Since I show a maximum of the first 1000 elements of the array variable in the watch variable panel anyway, I decided to cheat and send the full size but only the values ​​of the first maximum 1000 elements. So there will still be information about the real size of the array, but it will be possible to see, as before, a maximum of 1000 first elements. This improves Debug mode performance when there are large array variables.

closed 2024-08-05 by prasimix