eez-open/studio #421: [Feature Request]: Support LVGL scrollbar mode in eez-studio

mverch67, 2024-06-16:
I didn't find a way to configure the scrollbar in lvgl projects.

```
- LV_SCROLLBAR_MODE_OFF : Never show the scrollbars
- LV_SCROLLBAR_MODE_ON: Always show the scrollbars
- LV_SCROLLBAR_MODE_ACTIVE: Show scroll bars while an object is being scrolled
- LV_SCROLLBAR_MODE_AUTO: Show scroll bars when the content is large enough to be scrolled
```

Reference:
https://docs.lvgl.io/master/overview/scroll.html#mode

mvladic, 2024-06-16:
Implemented.

In the Flags section there are two new properties:

[image]

If left empty then no call to `lv_obj_set_scrollbar_mode` and `lv_obj_set_scroll_dir` will be generated - this is default when new widget is added to the screen.

closed 2024-06-17 by prasimix

mverch67, 2024-06-19:
Works nicely! Thanks for your great support!