eez-open/studio #424: Support Button matrix widget

mvladic, 2024-06-20:
https://docs.lvgl.io/8.0/widgets/core/btnmatrix.html

liriyao, 2024-07-25:
Can eez studio set event for every single Button in the btnmatrix?

mvladic, 2024-07-25:
Use `VALUE_CHANGED` event. This event will send ID, i.e. index of the button pressed:

[image]

[image]

liriyao, 2024-07-25:
> Use `VALUE_CHANGED` event. This event will send ID, i.e. index of the button pressed: [...]

ths

closed 2024-09-23 by mvladic