eez-open/studio #425: Error Generating Bitmap in LVGL V9

elmonic, 2024-06-24:
hi
I'm using eez studio for LVGL v.9 project.
all pictures used in app not generated.
[image]
thanks
eez studio V0.14.0

mvladic, 2024-06-24:
https://github.com/eez-open/studio/issues/311#issuecomment-1986932418

mvladic, 2024-07-27:
> I am using Windows and according to your description, the entire conversion operation is performed by the software itself.

No action is required only for the point no. 3, you still need to do points no. 1 (install python 3) and 2 (install pypng and lz4 python modules).

elmonic, 2024-07-28:
Hello, thank you for your support.
I had not paid attention to the titles.
The problem was solved.

mvladic, 2024-09-20:
On the Mac, there is some additional issues. Here is what description from the [Discord](https://discord.com/channels/511131497801973761/511133306675396629/1230097334826958889)

> Ok, I think I found the problem: Try starting studio by going into the /Application/EEZ-Studio folder (right-click on the application and select 'Show package contents'), browse to Contents -> MacOS. Inside this directory there is a Unix executable called 'EEZ Studio'. Right click this one and select open. 
> This starts EEZ Studio and probably your build will work now.
>
> Why does this work: the python package adds the path to python in the .zprofile file which is not read if you start the application in the regular way.

closed 2024-09-27 by mvladic