eez-open/studio #426: [dashboard]The size of the widget in the docking manager container is not normal

dong-king, 2024-06-26:
**Describe the bug**
The size of the widgets in the docker container is not normal
The width and height of these two buttons are 80 and 40 respectively.
when running, the width and height of the two buttons are not 80 and 40.
The size of the two buttons is related to the layout of the docking manager container.

**Screenshots**
Edit mode:
[image]

Run mode:
[image]

**Desktop (please complete the following information):**
 - OS: win10
 - eez studio version: 0.14.2

mvladic, 2024-06-26:
Containers that are immediate children of the Docking Manager container are automatically resized to match available area. 
And consequently, all the child widgets of these containers are moved and resized proportionally by default. You can control these by using the following controls:

[image]

Above, I set button to pin to the top and left edge and also I set width and height to be fixed, so it will not be moved or resized when its parent change size.

[image]

mvladic, 2024-06-26:
"Pin to edge" and "Fix size" controls are also used for the case when "Scale to fit" option of the Page is checked

[image]

And for the User widgets if User widget is defined with some width and height, but on the page where it is used is set different width and height, so we use these controls to know how to resize widgets that are inside user widget.