eez-open/studio #433: Missing option on bitmaps

skypoldo, 2024-07-03:
**Describe the bug**
"Resize to Fit Bitmap" check box in widget property is not there.

**To Reproduce**
Insert a Bitmap widget on a screen. When selecting a bitmap file larger than the widget there's no option to resize it to fit the widget.

**Expected behavior**
Seems like it's not possible to resize the image to the widget size anymore.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 10
- EEZ Studio 0.14.2

** LVGL version (if used)**
- 9.x

**Additional context**
Has this feature been removed? It's a very helpful one! Compared to the documentation the current version adds pivot points, scale and rotation (that's great, by the way!)
Thank you for the great program.

mvladic, 2024-07-10:
There is a "Resize to Fit Bitmap" button for Bitmap widget in Dashboard and EEZ-GUI projects, but not for the LVGL projects. And we never had that option. BTW in LVGL projects widget is called Image not Bitmap. If Width and Height for the Image widget is set to Content then widget will be automatically resized to fit image:

[image]

If Width and Height is set to px and width and height is greater than image width and height then in version 8.3 it will be displayed as so called [Mosaic](https://docs.lvgl.io/8.4/widgets/core/img.html#mosaic):

[image]

In version 9.0 for this case there is a new option called [inner align](
[image]
). We should definitely support this option.

mvladic, 2024-07-10:
I've added Inner align property in Image widget for LVGL version 9 #446

skypoldo, 2024-07-10:
Thank you for the clarification, Martin. Looking forward to seeing further development.

closed 2024-07-10 by skypoldo