eez-open/studio #435: [LVGL] Add support for all the existing widgets in LVGL version 8.4 and 9.1

mvladic, 2024-07-05:
Add support for the following widgets:

- [ ] Animation Image
- [ ] Button Matrix
- [ ] Canvas
- [ ] Image button **(only in 9.1, there is already for the 8.4)**
- [ ] Line
- [ ] LED
- [ ] List
- [ ] Lottie **(only in 9.1)**
- [ ] Menu
- [ ] Message box
- [ ] Span
- [ ] Spinbox
- [ ] Table
- [ ] Tabview
- [ ] Tile view
- [ ] Window

jcuypers, 2024-07-06:
@mvladic What about meter?   The eez studio example currently only works on 8.x.   I agree there is scale, but it lacks many of the 'specifics' configuration parameters as seen in the lvgl examples.  Also meter seems much more flexible as compared to scale (I might be wrong

jcuypers, 2024-07-06:
@mvladic Another component missing is colorwheel

mvladic, 2024-07-06:
Unfortunately, LVGL in version 9.x doesn't have Meter widget. I was surprised when I learned about this since it is really nice widget.

0xjmux, 2024-07-11:
Hi mvladic - it's really great to see that you're adding support for all the LVGL widgets! If it's not too much trouble, could I ask if you could move the "Tabview" widget up the queue a bit? EEZ-studio is sick and I really want to try it out for a personal project I'm doing, but because tabview isn't supported I've been attempting to design my UI by hand (which has been rough 😭). Thanks!

mvladic, 2024-07-11:
Tabview widget is already added in the master branch.

0xjmux, 2024-07-14:
that's awesome, thank you!