eez-open/studio #436: [lvgl] gesture event LV_EVENT_GESTURE missing

lehoangphuc86, 2024-07-06:
Hi Team, 

I am new to this tool. While designing our GUI, it seems we are missing the LV_EVENT_GESTURE event on the event dropdown.
May I know how to detect the gesture event like the lvgl natively supports?

Appreciate your response.
Version: 0.14.2
LVGL: 8.3

Peter

mvladic, 2024-07-06:
Done. Will be released in EEZ Studio version 0.15.0.

Here is an example project:

[test_gesture_event.zip](https://github.com/user-attachments/files/16116789/test_gesture_event.zip)

[image]

mvladic, 2024-07-06:
To receive Gesture event, "Gesture bubble" flag must be unchecked:

[image]

lehoangphuc86, 2024-07-06:
Thanks for your quick action. It means there is no option to get the gesture event in 0.14.2 right?
Also, I am working in lvgl + eez without flow. In my screen, I only have the dropdown to select the possible events.

[image]

mvladic, 2024-07-07:
> It means there is no option to get the gesture event in 0.14.2 right?

Correct.

mvladic, 2024-07-07:
> Also, I am working in lvgl + eez without flow. In my screen, I only have the dropdown to select the possible events.

You can always add event handler from the code.

lehoangphuc86, 2024-07-07:
> > Also, I am working in lvgl + eez without flow. In my screen, I only have the dropdown to select the possible events.
> 
> You can always add event handler from the code.

It means, I need to modify the generated source code to handle the LV_EVENT_GESTURE event right?
I prefer to not touching the generated source code, because any time I modify something on the GUI, I will re-generate and it will override my modification. 
Currently, I am separating the generated source and the app implementation.
If you confirm the same, in the next version, I think it is better to add all supported events into the dropdown on the studio or let user add their customize event name .

Thank you. I will close the issue after your confirmation.

lehoangphuc86, 2024-07-08:
Thank you for the clarification. Please close this case.

closed 2024-07-08 by lehoangphuc86