eez-open/studio #439: [EEZ Studio / LVGL] Prevent the use of internal / reserved variables

jcuypers, 2024-07-08:
When defining variables in a project there should be some basic whether or not the variables are already used somewhere else in the project.

mvladic, 2024-07-12:
I'm not sure I understand what you're trying to say. Can you elaborate on this a bit?

jcuypers, 2024-07-13:
> I'm not sure I understand what you're trying to say. Can you elaborate on this a bit?

Yes sure, so if would choose/create a variable named "status" in my project, all works fine in the editor.  After generating the actual LVGL code, there are a lot of compiler errors since somewhere else in the framework this variable name is already used.

I know its a corner case and probably a lot of work to avoid, but its good to be known at least.