eez-open/studio #441: [LVGL with Flow] Properties doesn't change even if is checked, even with new widgets

petrubuciu, 2024-07-09:
**Describe the bug**
After a while eez studio v0.14.2 on my project (LVGL with flow) stop working properties on every widget(mean if i change doesn't apply, remain as before). so when i want to change a prop on a widget is not changing even it show that are checked

**To Reproduce**
Steps to reproduce the behavior:
i don't know how to reproduce this type of bug

**Expected behavior**
normal

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [MacOS]
 - Version [Sonoma 14.5 (23F79)]

** LVGL version (if used)**

- 8.3

**Additional context**
also i move the project to windows and works to change the properties on widgets, but on my macos i can't, i even try to reinstall the eez studio.

There is a method to reset completly the eez studio? like refresh install? thanks

mvladic, 2024-07-09:
Does this always happen all the time when you start Studio on a Mac? I mean, does it happen immediately after starting the Studio or after some time of using the Studio?

petrubuciu, 2024-07-10:
yes, happend immediatly after i start Studio on my Mac, but after i delete all files from Studio from folder library works again. also i don't know how to reproduce this bug.

mvladic, 2024-07-10:
If this happens again please make a copy of files you delete from Studio from folder library and send it to me, this would help me to diagnose the problem. Also, do you have arm or intel mac?

petrubuciu, 2024-07-10:
ok, if this happen again i will send you the folder. I have mac with arm