eez-open/studio #451: LVGL Native - Generated Dropdown widget prevented from selecting other options with encoder

0xjmux, 2024-07-14:
**Describe the bug**
In Native LVGL, when you try to use a dropdown with a rotary encoder you are unable to change the value from what the application thinks the value should be via `get_var_xx()`. Trying to scroll the dropdown results in the focus immediately snapping back to the stored value, and you're unable to navigate the items in the dropdown to select a new one.

If this is something I'm doing wrong and not a bug, please let me know. 

#### To Reproduce
Steps to reproduce the behavior:
1. Create Project based on "native-interface-lvgl-no-flow" example
2. Integrate encoder input device with LVGL
3. Flash to device
4. Select 'dropdown' widget using rotary encoder
5. Attempt to scroll to other items in the dropdown. When you navigate off of the current selection, the interface immediately snaps you back to the original value.

**Expected behavior**
Scroll to select an entry in the dropdown like you can in the [LVGL keypad/encoder demo](https://github.com/lvgl/lvgl/tree/master/demos/keypad_encoder)

#### Versions
* LVGL v8.3
* EEZ-Studio built from source from commit b1606bf8  (July 13, 2024)
* OS: Kubuntu 22.04LTS

#### Additional Context

Based on my debugging, I'm pretty sure this is a bug related to EEZ-studio code generation not being optimized for rotary encoder use. The [official LVGL Encoder demo](https://github.com/lvgl/lvgl/tree/master/demos/keypad_encoder) has expected behavior with the dropdown and the encoder, it's just the EEZ studio generated code that doesn't. I've tried as many combinations of flags on the widget that I can think of, but nothing seems to work.

As far as I can tell, this has to do with how the EEZ-studio generated code sets variables in `screens.c`. **The code from below is the generated code, not from the files shown in EEZ-Studio**

* `screens.c` - This shows how the dropdown value is updated in the generated code. The current UI value is compared to the application value, and if they differ the UI value is changed to reflect the application state. This would explain why when you try to scroll in the dropdown it immediately snaps you back to the last saved value, and doesn't let you move to select a new one.

```c
void tick_screen_home() {
    {
        int32_t new_val = get_var_grinder_id();
        int32_t cur_val = lv_dropdown_get_selected(objects.obj2);
        if (new_val != cur_val) {
            tick_value_change_obj = objects.obj2;
            lv_dropdown_set_selected(objects.obj2, new_val);
            tick_value_change_obj = NULL;
        }
    }
```

For reference, the relevant `vars.c`: 
```c
static int32_t ui_grinder_id = 0;

int32_t get_var_grinder_id() {
        return ui_grinder_id;
}

void set_var_grinder_id(int32_t value) {
    ui_grinder_id = value;
}
```

#### Solution
I'm not familiar enough with LVGL or the EEZ-studio codebase to be sure. Based on my understanding of the issue, ideally the generated code would not keep updating the index while the widget is focused and you're selecting a new value. I'm guessing this issue also applies to widgets like 'roller' that work similarly.

mvladic, 2024-07-14:
Fixed.

0xjmux, 2024-07-15:
That was fast! The fix works perfectly, thank you!

closed 2024-07-15 by 0xjmux