eez-open/studio #453: eez delete me all flow, also local var, global var define as NATIVE

petrubuciu, 2024-07-16:
**Describe the bug**
when i open the eez studio v0.14.2 to continue my work, i saw that i have 50 error in checks. also when i looked at my global vars i saw that are [NATIVE]name_of _var but all globals was not define as native. also i can't find any flow that i have created. and local vars on each screen created deleted i think because in checks tells me that local var doesn't exists

**To Reproduce**
Steps to reproduce the behavior:
1. just fresh open the eez studio

**Expected behavior**

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS (arm version M1 max)
 - Version Sonoma 14.5

** LVGL version 8.3**

**Additional context**
i dont know if i was doing something wrong, yesterday i change from lvgl 8.3 to 9 and then i export(build) also then i switched back to 8.3. but even after that i switched back to 8.3 i was able to export(build) the project with no errors. but today was not working at all

mvladic, 2024-07-16:
Is it possible that you unchecked "Flow supprort" switch?

[image]

petrubuciu, 2024-07-16:
is active, also export export eez flow engine

mvladic, 2024-07-16:
This could happen if at one point flow support is disabled and then again enabled (unfortunately, if you saved your project after this all your flow specific stuff will be lost). Indication for this is:

> ... when i looked at my global vars i saw that are [NATIVE]name_of _var but all globals was not define as native" ...

When you disable flow support all your variables will become native.

mvladic, 2024-07-16:
Of course, I don't want to claim that you did something like this. I'm just trying to figure out what could have happened, i.e. in which scenario something like that (all global variables becoming native) could happen.

petrubuciu, 2024-07-16:
i dont remember if i diesable flow support. its posible. there is a posibile solutie to get back? or i need to start again the flows? :(

mvladic, 2024-07-16:
If you saved your project after you disabled flow support, all the flow specific parts of the project are lost.

mvladic, 2024-07-16:
I'm really sorry if your work has been lost 😒I hope it will not be too much of the effort to create it again, maybe this time will be even better. I created an issue #454 so that user is at least alerted of the irreversibility of this action.

closed 2024-09-27 by mvladic