eez-open/studio #456: Allow enabling default LVGL dark theme in editor and generated files

tylercamp, 2024-07-25:
**Is your feature request related to a problem? Please describe.**
LVGL offers a default dark theme which I'd like to use.

**Describe the solution you'd like**
I'd like a checkbox in the project settings page for whether to use light or dark theme for the project. It would modify `screens.c` code generation and apply to widgets shown within EEZ Studio editor

**Describe alternatives you've considered**
I'm currently applying a new theme with dark mode after the init call to EEZ Studio's generated code, but I'd like to manage the setting and see a preview of those changes directly in EEZ Studio.

---

The code-gen update would be easy but I'm not sure about updating widget styling within EEZ Studio's preview. I see a reference to a [wasm](https://github.com/eez-open/studio/blob/8fa02efb7f30b6f203bed2bdb15cd6e325402a5d/packages/project-editor/lvgl/widgets/Button.tsx#L88) wrapper util, so I assume there's some sort of LVGL web binding being used, but I can't tell how the [lvgl_runtime_*](https://github.com/eez-open/studio/tree/master/packages/project-editor/flow/runtime) files are being generated

(If it's using a proper LVGL API binding then updating EEZ UI for dark mode would probably be easy too, but I can't tell whether that's the case)

mvladic, 2024-07-26:
Implemented.

[image]

closed 2024-08-05 by prasimix

gwiethaus, 2024-09-02:
First, please excuse my English

In the new version EEZ Studio 0.16.0 I enabled the "dark theme" and it only changed the development screen. The editor continues with the default white theme.

mvladic, 2024-09-02:
For the EEZ Studio you can select dark theme here:

[image]

gwiethaus, 2024-09-02:
Thanks for the tip.
I created a new database. However, I discovered that I can't access the bottom of the screen. I think the scroll bar to reach the rest of the features is missing. I marked it in red as far as I can see on the screen. Remember that I'm using a laptop.

[image]

gwiethaus, 2024-09-02:
> For the EEZ Studio you can select dark theme here: [...]

Thanks for the tip.
I created a new database. However, I discovered that I can't access the bottom of the screen. I think the scroll bar to reach the rest of the features is missing. I marked it in red as far as I can see on the screen. Remember that I'm using a laptop.

[image]

mvladic, 2024-09-02:
Hm, this is bug. Fortunately, you can also select dark theme from the main menu:

[image]