eez-open/studio #460: [LVGL] Allow animations for sliders and bars

zalexzperez, 2024-07-26:
Currently, sliders and bars change values instantly, without an animation.
LVGL provides an ON/OFF animation parameter in their respective functions (lv_slider_set_value & lv_bar_set_value)

Please, allow enabling animations from the editor, as well as setting their duration, and if possible, the animation type too.

mvladic, 2024-07-27:
Implemented for Slider and Bar widget. Animation duration (in version 8.x it is called Animation time) is set in styles:

[image]

I didn't find a way to set animation type.

zalexzperez, 2024-07-27:
Great, thanks!
Just a thought, doesn't the "Enable animation" switch feel a bit redundant? It seems like the "Anim duration" checkbox could handle turning animations on and off by itself: 
-if left unchecked, animations are off. 
-If checked, animation duration could be enforced to be larger than 0,  in case it's required by LVGL.

mvladic, 2024-07-27:
The same redundancy exists in LVGL:

- There is a third parameter in `lv_slider_set_value` & `lv_bar_set_value` function calls where you can enable/disable animation with LV_ANIM_ON/LV_ANIM_OFF.
- If animation is enabled then duration of animation is controlled with the LV_STYLE_ANIM_DURATION

So, even if you send LV_ANIM_ON to `lv_slider_set_value` & `lv_bar_set_value` you will not see animation if LV_STYLE_ANIM_DURATION is set to 0; and otherwise, if you send LV_ANIM_OFF you will not see animation event if LV_STYLE_ANIM_DURATION is greater than 0.

We always try to follow the logic of LVGL.

zalexzperez, 2024-07-28:
I was editing my previous message to comment something I discovered after actually trying it, so I'll post it here.

Enabling and setting a duration time for the Anim duration doesn't do anything for Indicator and Knob parts of the Slider widget (same behavior from Bars expected). In other words, the animation only works when time is set in the Main part!

Now, replying to your previous comment, I see what you mean. 
The redundancy in the current implementation replicates what is possible to do with the given functions by LVGL.
But, if you end up only leaving the animation duration in the Main part, and remove it elsewhere, the master switch will have little use other than quickly turn off/on the already configured animations (those with duration > 0). On the other hand, removing the switch removes an extra step to get a real animation happening and a source of confusion for some distracted editors ("why does the slider have no animation?!").

closed 2024-07-29 by zalexzperez

liriyao, 2024-12-23:
how to use these setting ,any examples?