eez-open/studio #462: [Dashboard] 'requestAnimationFrame' handler took <N>ms and memory leak

dong-king, 2024-07-27:
**Describe the bug**
when i run [JSON.eez-project](https://github.com/eez-open/eez-project-examples/blob/master/examples/Dashboard/JSON.eez-project) in `eez-studio exmaple`, the messages('requestAnimationFrame' handler took <N>ms) are continuously output in DevTools.

And there is a memory leak in this example. when running this example for a while, the memory usage keeps rising.

I feel that the message's generation is related to the defined `CountryCity` structure.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: win10
 - Version: eez-studio v0.14.2

mvladic, 2024-07-27:
I didn't notice any memory leak wile running this dashboard for about 30 minutes. Memory can go up for a period of time, but it will go down again after garbage collector cycle.

As for this warning message, it's a performance issue and I've made some optimizations so it should run faster now and probably won't show this warning anymore.

dong-king, 2024-07-28:
It is not a memory leak.
The memory usage keeps increasing because DevTools has been open, and the memory it occupies continues to rise.