eez-open/studio #464: [LVGL Native] Generated UIs have several "ghost entries" when navigated by Rotary Encoder

0xjmux, 2024-07-30:
**Describe the bug**
When navigating my EEZ-Studio generated UI using a rotary encoder, there are several clicks where no item on screen is highlighted, and the cursor seems to just disappear. This seems to take place when wrapping around from the last entry in the widgets structure to the first. 

I've tried removing all 'scrollable' and 'clickable' attributes from any items meant to not be selectable (which is another issue - those attributes seem to reset themselves back to defaults every few saves) - but even with all those attributes removed, there are still 4-6 'clicks' of the rotary encoder (which correspond to [LV_KEY_RIGHT](https://docs.lvgl.io/7.11/porting/indev.html#other-features) ) where the cursor disappears from the screen. 

**To Reproduce**
Steps to reproduce the behavior:
Steps to reproduce the behavior:
1. Create Project based on "native-interface-lvgl-no-flow" example
2. Integrate encoder input device with LVGL
3. Flash to device
4. Between last element in the widgets structure and the first, the on-screen selection will disappear for several clicks (usually between 4-6).

**Expected behavior**
I expect the cursor/selection to wrap from the last element in the widgets list to the first, without 'getting lost' for several items in between - i've seen it be very confusing for others when the cursor just disappears from the screen for several clicks.

**Desktop (please complete the following information):**
Kubuntu 22.04LTS
EEZ studio built from master, commit `caaca947`

** LVGL version (if used)**

- 8.3.0

**Additional context**
Add any other context about the problem here.

mvladic, 2024-07-31:
I don't have any device with rotary encoder so I can't test it with the device. I can test it in the simulator where mouse wheel simulates rotary encoder. With mouse wheel I can select widget and when I press the mouse wheel button i can execute an action on selected widget. When testing like this everything works as expected.

Some points:

- If you called `lv_group_set_default(my_group)` then all widgets will be automatically added to the group.
- You can remove all the widgets from the group with the `lv_group_remove_all_objs(my_group)` and add particular widget to the group with the `lv_group_add_obj(my_group, objects.my_widget)` - this way you can control what's inside the group.

Currently, Studio doesn't support groups so you must do it in your code.

> ... (which is another issue - those attributes seem to reset themselves back to defaults every few saves)

This should be fixed now in master.

0xjmux, 2024-07-31:
Yeah, I have the following wrapper code to instantiate the UI and get things set up (which is mostly from the esp-idf lvgl example)

```c
void setup_lv_display() {

    /***** LVGL SETUP  *******/
    static lv_disp_draw_buf_t disp_buf; // contains internal graphic buffer(s) called draw buffer(s)
    static lv_disp_drv_t disp_drv;      // contains callback functions

    ESP_LOGI(TAG, "Turn off LCD backlight");
    gpio_config_t bk_gpio_config = {
        .mode = GPIO_MODE_OUTPUT,
        .pin_bit_mask = 1ULL << LCD_LED_PIN
    };
    ESP_ERROR_CHECK(gpio_config(&bk_gpio_config));

    ESP_LOGI(TAG, "Initialize SPI bus");
    spi_bus_config_t buscfg = {
        .sclk_io_num = LCD_SPI_SCK,
        .mosi_io_num = LCD_SPI_MOSI,
        .miso_io_num = LCD_SPI_MISO,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = LCD_H_RES * 80 * sizeof(uint16_t),
    };
    ESP_ERROR_CHECK(spi_bus_initialize(LCD_HOST, &buscfg, SPI_DMA_CH_AUTO));

    ESP_LOGI(TAG, "Install panel IO");
    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_spi_config_t io_config = {
        .dc_gpio_num = LCD_SPI_DC,
        .cs_gpio_num = LCD_SPI_CS,
        .pclk_hz = LCD_PIXEL_CLK_HZ,
        .lcd_cmd_bits = LCD_CMD_BITS,
        .lcd_param_bits = LCD_PARAM_BITS,
        .spi_mode = 0,
        .trans_queue_depth = 10,
        .on_color_trans_done = example_notify_lvgl_flush_ready,
        .user_ctx = &disp_drv,
    };
    // Attach the LCD to the SPI bus
    ESP_ERROR_CHECK(esp_lcd_new_panel_io_spi((esp_lcd_spi_bus_handle_t)LCD_HOST, &io_config, &io_handle));

    esp_lcd_panel_handle_t panel_handle = NULL;
    esp_lcd_panel_dev_config_t panel_config = {
        .reset_gpio_num = LCD_RST_PIN,
        .rgb_ele_order = LCD_RGB_ELEMENT_ORDER_BGR,
        .bits_per_pixel = 16,
    };
    ESP_LOGI(TAG, "Install ILI9341 panel driver");
    ESP_ERROR_CHECK(esp_lcd_new_panel_ili9341(io_handle, &panel_config, &panel_handle));

    ESP_ERROR_CHECK(esp_lcd_panel_reset(panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_init(panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_swap_xy(panel_handle, true));
    ESP_ERROR_CHECK(esp_lcd_panel_mirror(panel_handle, true, true));

    // user can flush pre-defined pattern to the screen before we turn on the screen or backlight
    ESP_ERROR_CHECK(esp_lcd_panel_disp_on_off(panel_handle, true));

    ESP_LOGI(TAG, "Initialize LVGL library");

    lv_init();
    // alloc draw buffers used by LVGL
    // it's recommended to choose the size of the draw buffer(s) to be at least 1/10 screen sized
    lv_color_t *buf1 = heap_caps_malloc(LCD_H_RES * 20 * sizeof(lv_color_t), MALLOC_CAP_DMA);
    assert(buf1);
    lv_color_t *buf2 = heap_caps_malloc(LCD_H_RES * 20 * sizeof(lv_color_t), MALLOC_CAP_DMA);
    assert(buf2);
    // initialize LVGL draw buffers
    lv_disp_draw_buf_init(&disp_buf, buf1, buf2, LCD_H_RES * 20);

    ESP_LOGI(TAG, "Register display driver to LVGL");
    lv_disp_drv_init(&disp_drv);
    disp_drv.hor_res = LCD_H_RES;
    disp_drv.ver_res = LCD_V_RES;
    disp_drv.flush_cb = example_lvgl_flush_cb;
    disp_drv.drv_update_cb = example_lvgl_port_update_callback;
    disp_drv.draw_buf = &disp_buf;
    disp_drv.user_data = panel_handle;
    lv_disp_t *disp = lv_disp_drv_register(&disp_drv);

    ESP_LOGI(TAG, "Install LVGL tick timer");
    // Tick interface for LVGL (using esp_timer to generate 2ms periodic event)
    const esp_timer_create_args_t lvgl_tick_timer_args = {
        .callback = &example_increase_lvgl_tick,
        .name = "lvgl_tick"
    };
    esp_timer_handle_t lvgl_tick_timer = NULL;
    ESP_ERROR_CHECK(esp_timer_create(&lvgl_tick_timer_args, &lvgl_tick_timer));
    ESP_ERROR_CHECK(esp_timer_start_periodic(lvgl_tick_timer, EXAMPLE_LVGL_TICK_PERIOD_MS * 1000));

   // call encoder setup
    lv_indev_t *indev_encoder = encoder_indev_init();

    // setup group
    lv_group_t *group = lv_group_create();
    lv_group_set_default(group);
    lv_indev_set_group(indev_encoder, group);

    /***** END LVGL SETUP  *******/
}
```

I'm using the default groups to do navigation since adding everything manually would be a pain - I can give doing it manually a try as well. 

Will do some triage to see if I can narrow down exactly what in the generated code is causing the issue. I'm currently unable to use the simulator with my UI - not sure if that's a native LVGL rendering issue or something else, but whenever I click on "run" it just shows me a blank screen - so manual it is.

> This should be fixed now in master.

Awesome, thanks!

mvladic, 2024-07-31:
> Will do some triage to see if I can narrow down exactly what in the generated code is causing the issue. I'm currently unable to use the simulator with my UI - not sure if that's a native LVGL [...]

You said that you are using EEZ studio built from master. How do you build it?

Can you send me your project file, so I can try to Run it in simulator?

0xjmux, 2024-07-31:
* I've been doing `git pull`, followed by `npm run build`, and then `npm start`. It's been working fine in the past, but just now got issues when building along the lines of:

```
packages/project-editor/store/paste-with-dependencies.tsx(427,43): error TS2345: Argument of type 'PasteObject | undefined' is not assignable to parameter of type 'PasteObject'.
  Type 'undefined' is not assignable to type 'PasteObject'.
packages/project-editor/store/paste-with-dependencies.tsx(428,36): error TS2345: Argument of type 'PasteObject | undefined' is not assignable to parameter of type 'PasteObject'.
  Type 'undefined' is not assignable to type 'PasteObject'.
```

Can email you a copy of my .eez-project file, since my project isn't public atm.

mvladic, 2024-07-31:
> I've been doing git pull, followed by npm run build, and then npm start. It's been working fine in the past, but just now got issues when building along the lines of:

This is my fault, sorry. I just pushed a fix.

> Can email you a copy of my .eez-project file, since my project isn't public atm.

Thanks. You can find my email here: https://github.com/mvladic. I will delete the project as soon as I'm done with debugging.

0xjmux, 2024-07-31:
`npm run build` is still failing on 68a3ec42, with the same error. 

Attempting to start the application fails with 
```
(node:443023) electron: Failed to load URL: file:///[PATH]/eez-studio/studio/build/home/index.html with error: ERR_FILE_NOT_FOUND
electron: Failed to load URL: file:///[PATH]/eez-studio/studio/build/home/index.html with error: ERR_FILE_NOT_FOUND
[443057:0731/082634.939943:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 1 times!
[443057:0731/082641.258898:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 2 times!
```

mvladic, 2024-07-31:
I think you have old TypeScript compiler. Try `npm install` before `npm run build`, it should install a newer version.

0xjmux, 2024-08-02:
that was it, was able to build and run successfully.

will do that root cause triage and let you know what i find as soon as i'm able to.

closed 2024-09-27 by mvladic