eez-open/studio #465: [LVGL Native] Arc widget unable to be selected by rotary encoder navigation

0xjmux, 2024-07-30:
**Describe the bug**
I have an LVGL 'Arc' widget in my EEZ-studio generated UI, and I want to be able to use my rotary encoder input device to interact with it. Currently, the cursor skips right over it in the menu. No matter what selection of Flags/attributes I apply to it, I'm unable to change this behavior. 

**To Reproduce**
Steps to reproduce the behavior:
Steps to reproduce the behavior:
1. Create Project based on "native-interface-lvgl-no-flow" example. Add an 'Arc' Widget to the UI, using the 'flex' layout to make navigating between items work as expected;
2. Integrate encoder input device with LVGL
3. Flash to device
4. Try to select the 'arc' widget to modify the value. The widget is skipped over by the cursor, and you are unable to select it.

**Expected behavior**
I expected to be able to focus on the Arc widget with my rotary encoder navigation, select it, and then use the rotary encoder to change the value like is possible in the LVGL encoder example.

**Desktop (please complete the following information):**
Kubuntu 22.04LTS
EEZ studio built from master, commit `caaca947`

** LVGL version (if used)**
- 8.3.0

**Additional**
Thanks for your work with EEZ-studio, it's a really great tool - much better than having to build my UI from scratch :)

mvladic, 2024-07-31:
Unfortunately, Arc widget can not be added to the group thus can't be focused with the encoder! This is limitation of LVGL.

I just checked the source code of the LVGL and found out that only the widgets with the:

```
.group_def = LV_OBJ_CLASS_GROUP_DEF_TRUE,
```
in the class definition can be added to the group and be in focus.

Here is a class definition for the Arc:

```
const lv_obj_class_t lv_arc_class  = {
    .constructor_cb = lv_arc_constructor,
    .event_cb = lv_arc_event,
    .instance_size = sizeof(lv_arc_t),
    .editable = LV_OBJ_CLASS_EDITABLE_TRUE,
    .base_class = &lv_obj_class
};
```

As you can see `.group_def = LV_OBJ_CLASS_GROUP_DEF_TRUE` is missing.

And here is, for example, a class definition for the Slider:

```
const lv_obj_class_t lv_slider_class = {
    .constructor_cb = lv_slider_constructor,
    .event_cb = lv_slider_event,
    .editable = LV_OBJ_CLASS_EDITABLE_TRUE,
    .group_def = LV_OBJ_CLASS_GROUP_DEF_TRUE,
    .instance_size = sizeof(lv_slider_t),
    .base_class = &lv_bar_class
};
```
Here, there is a ` .group_def = LV_OBJ_CLASS_GROUP_DEF_TRUE,` so Slider can be added to the group and be in focus.

0xjmux, 2024-07-31:
Opened an issue in the LVGL repo for this, hopefully it can get fixed there and then propagated upstream

mvladic, 2024-07-31:
https://github.com/lvgl/lvgl/issues/6609