eez-open/studio #469: Add LVGL key definitions to expression builder.

fietser28, 2024-07-31:
**Is your feature request related to a problem? Please describe.**
When using key events from LVGL in flow you have to manually lookup the key codes and hardcode them. 

**Describe the solution you'd like**
Add the LVGL_KEY_* to the expression builder. In the generated code these constants are already present. This is very similar to the LVGL symbols already in the expression builder.

**Describe alternatives you've considered**
Hardcoding the values. You also first have to look them up in LVGL documentation.

mvladic, 2024-07-31:
Actually, we should add system enumeration type $LVGLKey like we already have $LVGLDir.

mvladic, 2024-07-31:
[image]

mvladic, 2024-07-31:
Hm, on second thought I don't know why I added System enumeration $LVGLDir when I already have prior case with LVGL.LV_SYMBOL_....

fietser28, 2024-07-31:
From a UI perspective it is nice if the list like LVGLDir, SYMBOL and KEYS can collapse. 
With increasing features some of the lists become long to scroll.

Strictly speaking the KEYS in LVGL are en enumeration to but in the UI there is no need difference between constants and enumerations.

mvladic, 2024-07-31:
I will add $LVGLKey enumeration for now. And later we can improve Expression Builder UI so LVGL.LV_SYMBOL_... are collapsible.

mvladic, 2024-07-31:
Implemented.

[image]

fietser28, 2024-07-31:
Working fine. Thank you!

closed 2024-07-31 by fietser28