eez-open/studio #473: [Dashboard] throwError: Execution queue is full

dong-king, 2024-08-06:
**Describe the bug**
When i run my dashboard project, an error is thrown: Execution queue is full.

[image]

Is this because I use too many widget and action components in my project?

mvladic, 2024-08-06:
Yes, the reason could be that you have too many widgets. Or some logical error inside your flow. For example, if you have some loop in your flow where inside the loop there is an action which has two outputs and execution continue from both outputs simultaneously and execution from these outputs both run in a loop - basically doubles execution flow for each loop iteration execution. The first reason (too many widget) can be easily fixed and will be fixed in the HF version. The second reason must be fixed in the flow.

mvladic, 2024-08-06:
This is an example of the flow that will inevitably result in "Execution queue is full":

[image]

mvladic, 2024-08-06:
Max. queue size for dashboard is increased from 1000 to 100000.

closed 2024-09-27 by mvladic