eez-open/studio #475: [LVGL] No settings for spinbox

elmonic, 2024-08-06:
hi
Spinbox is one of the most used widgets in graphic systems. It will be great if you add settings.
As shown below:
[image]
Same as the settings below
[image]

mvladic, 2024-08-06:
Do you know how "Increment" is used?

mvladic, 2024-08-06:
Implemented.

[image]

Here is some sample project:

[spinbox_test.zip](https://github.com/user-attachments/files/16509106/spinbox_test.zip)

[image]

elmonic, 2024-08-07:
> Implemented. [...]

Hi, thanks
Yes, it can be done with flow mode.
Will this section be added in the next version?

mvladic, 2024-08-07:
This will be released very soon in version 0.15.1

closed 2024-09-27 by mvladic